JuulLabs/kable #789: [Feedback: 0.35.0-RC]: Scanner doesn't show in its description that the InternalException can be thrown

PavlosTze, 2024-11-12:
Based on #733:

> All scanning operations (and requestDevice on JavaScript) only throw the following exceptions:
> 
> - IllegalStateException — thrown if a static state prevents scanning (e.g. bluetooth unavailable, scanning feature not supported or insufficient permission) — essentially, anything that could've/should've been checked prior to initiating the scan
> - UnmetRequirementException (extends IOException) — thrown if a transient state prevented scanning (e.g. bluetooth disabled) — essentially, anything that even if you check it ahead of time, may not be satisfied by the time the scan is initiated
> - InternalException — if a condition was hit that was unexpected (this is a bug and exception message asks consumers to report when this exception is raised)

But hovering over the respective functionality doesn't report the `InternalException`:
[image]

twyatt, 2024-11-13:
I actually don't know what the best approach is for this.

`InternalException` is intended to only be thrown when a condition or state was hit that was not expected (i.e. we made a mistake in the implementation of Kable). As such, it likely is not a (gracefully) recoverable condition and probably shouldn't be caught. It would be expected that such an exception is allowed to propagate to crash the app, so that any crash reporting tools in the app could capture the failure so that it may be reported to Kable [issues](https://github.com/JuulLabs/kable/issues).

That being said, I worry that documenting the exception in functions will cause library consumers to catch the exception, which (as mentioned above) may put Kable in an inconsistent state, resulting in strange/unrelated failures that instead get reported (rather than the real issue originally propagated as `InternalException`.

Perhaps it is better for it to be documented but emphasis that it shouldn't be caught? At least that way, consumers are aware that it may be thrown, even if it shouldn't be handled.

PavlosTze, 2024-11-13:
Thanks for the explanation, I wasn't 100% aware of this situation tbh. In that case, I agree with your last paragraph, I think you should document it but emphasize that it shouldn't be caught or at least if it's gonna get caught, it should be reported somehow as an `InternalException`. 

For example in our case we can catch it, show a generic error to the user and by using we have planted `Timber.e` to the Crashlytics so it will be reported in our logs as well (so we can report it to you afterwards). Avoiding the crash but keeping/reporting the correct error.

So I think to avoid confusion of having exceptions not documented, it should be but explain why this should crash and not be allowed to continue after this exception.

PavlosTze, 2024-11-26:
Sounds great, thank you! I'm closing this issue.

PR #801 (2024-12-02): Change `InternalException` to `InternalError`
When an unrecoverable internal failure occurs, a sub-class of [`Error`](https://docs.oracle.com/javase/8/docs/api/?java/lang/Error.html) is more appropriate:

> indicates serious problems that a reasonable application should not try to catch

This is in line with [kotlinx.coroutines' approach](https://github.com/Kotlin/kotlinx.coroutines/blob/6c6df2b850382887462eeaf51f21f58bd982491d/kotlinx-coroutines-core/common/src/internal/DispatchedTask.kt#L131-L133), whereas `CoroutinesInternalError` is a sub-class of `Error` and also asks their library consum [... cut]
closed 2024-11-26 by PavlosTze