JuulLabs/kable #793: [Feedback `0.35.0-RC`]: Connecting after rebooting of the device produces a `java.lang.IllegalStateException: Scope is not active`

PavlosTze, 2024-11-12:
We have the following `connect` code in our codebase:
```
suspend fun connectToPeripheral(): Either<Failure, Unit> {
        return try {
            Timber.d("[BLE Communication]: Connecting to peripheral...")
            peripheral.connect()
            Either.Right(Unit)
        } catch (e: IllegalStateException) {
            ...
            Either.Left(BluetoothError.IllegalStateError())
        }
        ...
        ...
}
```

And the following flow:
1. Connect to a device
2. Write something to that device
3. Reboot the device and disconnect from it
4. Reconnect to the device after it has booted again
5. Get some information from that device

This flow works correctly with `0.34.0`. The problem is that with `0.35.0-RC` the `connect()` function throws an `IllegalStateException`:
`java.lang.IllegalStateException: Scope is not active`

I'm not sure what "Scope is not active" means here but ideally we would like for it to work as it did with `0.34.0` without having to re-write our codebase.

PavlosTze, 2024-11-13:
Thank you. Please ping me here when it's ready to test it again.

twyatt, 2024-11-22:
@PavlosTze The "scope is not active" exception message occurs when the peripheral's scope is no longer active but an operation was attempted using the scope (in this case, `connect`) — providing a better failure message will be tracked via #798.

This can happen if the call to `connect` was preceded by a call to `peripheral.cancel()`.
_Being that the peripheral's coroutine scope is a supervisor scope, it shouldn't shutdown unless explicitly instructed to do so, but there may be a bug._

I've added logging for when the peripheral's scope shuts down, so that we can try and find the cause; please try the following snapshot:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.35.0-rc+logging-SNAPSHOT")
}
```

When reproducing, be sure that the peripheral is configured with [logging enabled](https://github.com/JuulLabs/kable?tab=readme-ov-file#logging).

Then look for the following log messages (and associated stacktraces):
- `"Supervisor job ended"`
- `"Supervisor job received exception"`

PavlosTze, 2024-11-26:
I just did that, you can find the logs below. Also, based on your message, I removed the `peripheral.cancel()` from my disconnection code, and it worked perfectly, just like before. So I guess this might be an issue on my side, when I would expect a new automatic re-connection immediately after disconnecting, I should handle it in a better way and not call the `cancel()` function?

```
2024-11-26 10:04:01.236 25745-26018 BluetoothC...ionManager com.weatherxm.app                    D  [BLE Communication]: AT+BAND=5
2024-11-26 10:04:01.239 25745-26018 Kable/Peripheral        com.weatherxm.app                    D  XX:XX:XX:XX:36:7A write
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-8841-43f4-a8d4-ecbe34729bb3
                                                                                                    writeType: WithoutResponse
                                                                                                    data: 41 54 2B 42 41 4E 44 3D 35 0D 0A
2024-11-26 10:04:01.249 25745-26560 Kable/Callback          com.weatherxm.app                    D  F9:F8:40:77:36:7A onCharacteristicWrite
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-8841-43f4-a8d4-ecbe34729bb3
                                                                                                    status: GATT_SUCCESS(0)
2024-11-26 10:04:01.251 25745-26026 Kable/Peripheral        com.weatherxm.app                    D  XX:XX:XX:XX:36:7A Starting observation
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-1e4d-4bd9-ba61-23c647249616
2024-11-26 10:04:01.251 25745-26026 Kable/Peripheral        com.weatherxm.app                    V  XX:XX:XX:XX:36:7A setCharacteristicNotification
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-1e4d-4bd9-ba61-23c647249616
                                                                                                    value: true
2024-11-26 10:04:01.252 25745-26026 BluetoothGatt           com.weatherxm.app                    D  setCharacteristicNotification() - uuid: 49535354-1e4d-4bd9-ba61-23c647249616 enable: true
2024-11-26 10:04:01.253 25745-26026 Kable/Peripheral        com.weatherxm.app                    V  XX:XX:XX:XX:36:7A Writing ENABLE_NOTIFICATION_VALUE to CCCD
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-1e4d-4bd9-ba61-23c647249616
                                                                                                    descriptor: 00002902-0000-1000-8000-00805f9b34fb
2024-11-26 10:04:01.254 25745-26026 Kable/Peripheral        com.weatherxm.app                    D  XX:XX:XX:XX:36:7A write
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-1e4d-4bd9-ba61-23c647249616
                                                                                                    descriptor: 00002902-0000-1000-8000-00805f9b34fb
                                                                                                    data: 01 00
2024-11-26 10:04:01.384 25745-26560 Kable/Callback          com.weatherxm.app                    D  F9:F8:40:77:36:7A onCharacteristicChanged
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-1e4d-4bd9-ba61-23c647249616
                                                                                                    data: 0D 0A 4F 4B 0D 0A
2024-11-26 10:04:01.385 25745-26560 Kable/Callback          com.weatherxm.app                    D  F9:F8:40:77:36:7A onDescriptorWrite
                                                                                                    service: 49535354-fe7d-4ae5-8fa9-9fafd205e455
                                                                                                    characteristic: 49535354-1e4d-4bd9-ba61-23c647249616
                                                                                                    descriptor: 00002902-0000-1000-8000-00805f9b34fb
[... cut ...]
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:823)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:720)
                                                                                                    	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:707)
2024-11-26 10:04:22.432 25745-25745 ViewRootIm...mActivity] com.weatherxm.app
```

twyatt, 2024-11-26:
> when I would expect a new automatic re-connection immediately after disconnecting, I should handle it in a better way and not call the `cancel()` function?

Is it possible to share a snippet of your reconnection code, and I can give you feedback on it.
Feel free to DM me on kotlinlang Slack (`@travis`) if you'd prefer to share the code there.

PavlosTze, 2024-12-02:
> Is it possible to share a snippet of your reconnection code, and I can give you feedback on it. Feel free to DM me on [kotlinlang Slack](https://kotlinlang.org/community/) (`@travis`) if you'd [...]

Hey, sorry for the late reply. Our codebase is open source @ [WeatherXM/wxm-android](https://github.com/WeatherXM/wxm-android) so I've no issues sharing it here.

Regarding the flow: 

### Disconnection Code

It starts in the ViewModel at function [reboot](https://github.com/WeatherXM/wxm-android/blob/92c558341cfcbf62b1bd2a153b598cce5fa3d869/app/src/main/java/com/weatherxm/ui/claimdevice/helium/result/ClaimHeliumResultViewModel.kt#L91) ➡️➡️ [UseCase](https://github.com/WeatherXM/wxm-android/blob/feature/WIP-TEST-BLE-Library-Kable-0.35.0-rc/app/src/main/java/com/weatherxm/usecases/BluetoothConnectionUseCaseImpl.kt#L50) ➡️➡️ [Repository](https://github.com/WeatherXM/wxm-android/blob/feature/WIP-TEST-BLE-Library-Kable-0.35.0-rc/app/src/main/java/com/weatherxm/data/repository/bluetooth/BluetoothConnectionRepositoryImpl.kt#L50) (if you check the code here you will see that we need a custom fix of disconnecting from the peripheral also at this point) ➡️➡️ DataSource ([Reboot](https://github.com/WeatherXM/wxm-android/blob/feature/WIP-TEST-BLE-Library-Kable-0.35.0-rc/app/src/main/java/com/weatherxm/data/datasource/bluetooth/BluetoothConnectionDataSourceImpl.kt#L108) & [Disconnect](https://github.com/WeatherXM/wxm-android/blob/feature/WIP-TEST-BLE-Library-Kable-0.35.0-rc/app/src/main/java/com/weatherxm/data/datasource/bluetooth/BluetoothConnectionDataSourceImpl.kt#L56)) ➡️➡️ Bluetooth Connection Manager (the problem is here at the [disconnectFromPeripheral()](https://github.com/WeatherXM/wxm-android/blob/92c558341cfcbf62b1bd2a153b598cce5fa3d869/app/src/main/java/com/weatherxm/data/bluetooth/BluetoothConnectionManager.kt#L168) where we do `peripheral.cancel()`)

### Reconnection Code
It starts in the ViewModel at function [connectToPeripheral](https://github.com/WeatherXM/wxm-android/blob/feature/WIP-TEST-BLE-Library-Kable-0.35.0-rc/app/src/main/java/com/weatherxm/ui/claimdevice/helium/result/ClaimHeliumResultViewModel.kt#L107) ➡️➡️ [UseCase](https://github.com/WeatherXM/wxm-android/blob/92c558341cfcbf62b1bd2a153b598cce5fa3d869/app/src/main/java/com/weatherxm/usecases/BluetoothConnectionUseCaseImpl.kt#L26) ➡️➡️ [Repository](https://github.com/WeatherXM/wxm-android/blob/92c558341cfcbf62b1bd2a153b598cce5fa3d869/app/src/main/java/com/weatherxm/data/repository/bluetooth/BluetoothConnectionRepositoryImpl.kt#L26)➡️➡️ [DataSource](https://github.com/WeatherXM/wxm-android/blob/92c558341cfcbf62b1bd2a153b598cce5fa3d869/app/src/main/java/com/weatherxm/data/datasource/bluetooth/BluetoothConnectionDataSourceImpl.kt#L43) ➡️➡️ Bluetooth Connection Manager (the problem is here at the [connectToPeripheral()](https://github.com/WeatherXM/wxm-android/blob/92c558341cfcbf62b1bd2a153b598cce5fa3d869/app/src/main/java/com/weatherxm/data/bluetooth/BluetoothConnectionManager.kt#L190) which throws the above `IllegalStateException: java.lang.IllegalStateException: Scope is not active` exception)

twyatt, 2024-12-03:
Looking through the [WeatherXM/wxm-android](https://github.com/WeatherXM/wxm-android) code (if I'm understanding correctly), you only manage a single peripheral at a time (i.e. you can't connect to multiple weather stations concurrently)? If that is the case, then I would recommend only calling `Peripheral.cancel()` when the "managed" peripheral is changing. Essentially, just adding `peripheral?.cancel()` to the beginning of your `BluetoothConnectionManager.setPeripheral` function and removing your `peripheral.cancel()` call from your `BluetoothConnectionManager.disconnectFromPeripheral` function.

Basically, you should only _cancel_ a peripheral when you are done using it. `Peripheral.cancel` is synonymous with _disposing_ it.

Let me know if that seems feasible / works.

PavlosTze, 2024-12-06:
Yup you are 100% correct. I just tested it and it worked like a charm. Therefore I'm considering this issue as a bug in our side, so I'm closing it. Thank you very much for your help!

closed 2024-12-06 by PavlosTze