JuulLabs/kable #794: Peripheral.write() not sending request

chungchungdev, 2024-11-12:
- kable version: 0.35.0-rc
- platform: Android only

I have a device like the one below. The custom characteristic accepts write requests that allows reading, writing, subscribing and unsubscribing different data. The corresponding data is emitted through notification.
```kotlin
class MyDevice(private val peripheral: Peripheral) {

  private val customCharacteristic = characteristicOf("some-uuid1", "some-uuid2")

  private val myData: Flow<ByteArray> = peripheral.observe(customCharacteristic)

  val sensor1Flow = myData.map { /*filter sensor1 data*/ }
  
  val state1Flow = myData.map { /*filter state1 data*/ }

  lateinit var state1: String

  suspend fun subscribeSensor1() =
      peripheral.write(customCharacteristic, byteArrayOf(/*command1*/), WriteType.WithResponse)

  suspend fun readState1(): String {
    val deferred = peripheral.async { state1Flow.first() }
    peripheral.write(customCharacteristic, byteArrayOf(/*command2*/), WriteType.WithResponse)
    Timber.d("request sent")
    return deferred.await()
  }

  suspend fun connect() {
    peripheral.connect()
    subscribeSensor1()
    Timber.d("block?")
    state1 = readState1()
    Timber.d("didn't block")
  }
}
```
However, the `await()`  in `readState1()`  is blocking the coroutine scope. It seems that the request is not sent to the device. So no matching data is emitted from `state1Flow` and the coroutine scope is stuck forever. But when I add a line to collect any of the flow outside (e.g. `connect()` method or `viewmodelScope`), the peripheral scope is not blocked and it runs without problem. I discover this by chance but I don't know what makes the difference.
```kotlin
suspend fun connect() {
    myData.collect {} // or sensor1Flow or state1Flow
    peripheral.connect()
    subscribeSensor1()
    state1 = readState1()
  }
```

I also tried adding that line inside the method, but it is still blocking.
```kotlin
suspend fun readState1(): String {
    myData.collect {} // or sensor1Flow or state1Flow
    val deferred = peripheral.async { state1Flow.first() }
    peripheral.write(customCharacteristic, byteArrayOf(/*command2*/), WriteType.WithResponse)
    Timber.d("request sent")
    return deferred.await()
  }
```

twyatt, 2024-11-12:
Can you [provide logs](https://github.com/JuulLabs/kable?tab=readme-ov-file#logging), please?

chungchungdev, 2024-11-13:
I try to add a delay, it works. But how much time should be delayed depends on devices. My old device need more delay than my new device to work properly.
```kotlin
suspend fun readState1(): String {
    val deferred = peripheral.async { state1Flow.first() }
    delay(1000)
    peripheral.write(customCharacteristic, byteArrayOf(/*command2*/), WriteType.WithResponse)
    Timber.d("request sent")
    return deferred.await()
  }
```

twyatt, 2024-11-13:
Oh, if a `delay` is fixing it, then I suspect your `async` is not spinning up quickly enough.

You should use `onSubscription` instead of `async` for this, something to the effect of:

```kotlin
suspend fun readState1(): String =
    state1Flow
        .onSubscription {
            peripheral.write(customCharacteristic, byteArrayOf(/*command2*/), WriteType.WithResponse)
        }.first()
}
```

More details can be read [here](https://github.com/Kotlin/kotlinx.coroutines/issues/1758#issuecomment-626372261).

chungchungdev, 2024-11-13:
I tried this but it didn't work.
```kotlin
suspend fun readState1(): String =
    state1Flow
        .shareIn(peripheral, SharingStarted./*tried all of them*/)
        .onSubscription {
            peripheral.write(customCharacteristic, byteArrayOf(/*command2*/), WriteType.WithResponse)
    }.first()
```
I use `async(start = CoroutineStart.UNDISPATCHED)` as mentioned in your given link and it works! Thanks for the help..

twyatt, 2024-11-13:
For the `shareIn`, you'll want to start it prior to function invocation (so that it has ample time to spin up), for example:

```kotlin
private val incoming = state1Flow
    .shareIn(peripheral, SharingStarted.Eagerly)

suspend fun readState1(): String =
    incoming.onSubscription {
        peripheral.write(customCharacteristic, byteArrayOf(/*command2*/), WriteType.WithResponse)
    }.first()
```

Using `async(start = CoroutineStart.UNDISPATCHED)` should also work just fine. 👍

closed 2024-11-13 by twyatt