JuulLabs/kable #804: Connection maintained with incorrect password.

LuizAugusto1323, 2024-11-26:
After trying to connect with peripheral.connect() and requesting the product password, when I enter the wrong password, I get a response from Kable/Callback that the connection status has changed and the product has been disconnected, but then I receive a suppression from Kable/Observation about the connection failure and after that Kable/Peripheral tells me that I am connected, and I don't understand why this response is given.

BluetoothGatt           D  connect() - device: xx:xx:xx:xx:xx:1D, auto: false
BluetoothGatt           D  registerApp()
BluetoothGatt           D  registerApp() - UUID=xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
BluetoothGatt           D  onClientRegistered() - status=0 clientIf=7
BluetoothGatt           D  onClientConnectionState() - status=0 clientIf=7 device=xx:xx:xx:xx:xx:1D
Kable/Callback         D  xx:xx:xx:xx:xx:1D onConnectionStateChange,  status: Success, newState: Connected
Kable/Peripheral       V  xx:xx:xx:xx:xx:1D discoverServices
BluetoothGatt           D  discoverServices() - device: xx:xx:xx:xx:xx:1D
BluetoothGatt           D  onConnectionUpdated() - Device=xx:xx:xx:xx:xx:1D interval=6 latency=0 timeout=500 status=0
BluetoothGatt           D  onSearchComplete() = Device=xx:xx:xx:xx:xx:1D Status=0
Kable/Callback         D  xx:xx:xx:xx:xx:1D onServicesDiscovered, status: GATT_SUCCESS(0)
Kable/Peripheral       V  xx:xx:xx:xx:xx:1D Discovered 5 services
Kable/Peripheral       V  xx:xx:xx:xx:xx:1D Configuring characteristic observations
Kable/Peripheral       D  xx:xx:xx:xx:xx:1D setCharacteristicNotification, service: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx, characteristic: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx, value: true
BluetoothGatt           D  setCharacteristicNotification() - uuid: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx enable: true
Kable/Peripheral       V  xx:xx:xx:xx:xx:1D Writing ENABLE_NOTIFICATION_VALUE to CCCD,  service: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx, characteristic: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx,  descriptor: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
Kable/Peripheral       D  xx:xx:xx:xx:xx:1D write, service: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx,  characteristic: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx, descriptor: xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx,  data: 01 00
BluetoothGatt           D  onConnectionUpdated() - Device=xx:xx:xx:xx:7A:1D interval=39 latency=0 timeout=500 status=0

I entered the wrong password…

BluetoothGatt             D  onClientConnectionState() - status=22 clientIf=7 device=xx:xx:xx:xx:xx:1D
Kable/Callback           D  xx:xx:xx:xx:xx:1D onConnectionStateChange,  status: 22,  newState: Disconnected
BluetoothGatt             D  close()
BluetoothGatt             D  unregisterApp() - mClientIf=7
Kable/Observation     V  xx:xx:xx:xx:xx:1D Suppressed failure: com.juul.kable.ConnectionLostException
Kable/Peripheral         I  xx:xx:xx:xx:xx:1D Connected

twyatt, 2024-11-26:
By "password", are you referring to the PIN requested during pairing/bonding?
If so, bonding is not yet supported (tracked by #111).

LuizAugusto1323, 2024-11-27:
Yes, but in previous versions like 0.27.0, the return of Kable/Peripheral changes, instead of there being a suppression, the log tells me the following:

Kable/Callback  D  xx:xx:xx:xx:xx:1D onConnectionStateChange, status: Success, newState: Disconnected
BluetoothGatt   D  close()
BluetoothGatt   D  unregisterApp() - mClientIf=7
BluetoothGatt   D  close()
BluetoothGatt   D  unregisterApp() - mClientIf=0
BluetoothGatt   D  close()
BluetoothGatt   D  unregisterApp() - mClientIf=0
Kable/Peripheral  E  xx:xx:xx:xx:xx:1D Failed to connect,  kotlinx.coroutines.JobCancellationException: Job was cancelled; job=JobImpl{Cancelled}@cf40dac

With this return, the State of the peripheral returns Disconnected, in the return above, the state of the peripheral remains as connected, I imagine because of the suppression