JuulLabs/kable #814: Inconsistent behavior between native and filtered ManufacturerData

eygraber, 2024-12-06:
Using 0.35.0-rc

If I use one `Filter.ManufacturerData` and I pass a single element `ByteArray` for both `data` and `dataMask` and everything works fine (the actual manufacturer data is 9 bytes).

If I use multiple `Filter.ManufacturerData` and I pass a single element `ByteArray` for both `data` and `dataMask` I get a crash saying that:

> Data mask length (1) must match data length (9)

because `Filter.ManufacturerData.matches(data: ByteArray?)` calls `requireDataAndMaskHaveSameLength` using the actual manufacturer data (as opposed to the `data` parameter).

twyatt, 2024-12-10:
Thanks for reporting! Unfortunately I'm pretty swamped, so it may be a while before I can look into this.
PRs are also welcome.

eygraber, 2024-12-10:
I can send a PR, but I'm not sure what the correct behavior is supposed to be.

twyatt, 2025-01-20:
@eygraber when you have a chance, can you test if the following SNAPSHOT fixes the issue for you (it is based on [`0.35.0` release](https://github.com/JuulLabs/kable/releases/tag/0.35.0) with #842 cherry-picked):

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:kable-core:0.35.0+issue814-SNAPSHOT")
}
```

eygraber, 2025-01-20:
That fixes the issue for me, thanks!

PR #842 (2025-01-27): Allow for advertisement manufacturer data length to exceed filter data mask length
Closes #814

As seen on the ["Web Bluetooth / Manufacturer Data Filter Sample"](https://googlechrome.github.io/samples/web-bluetooth/manufacturer-data-filter.html?companyIdentifier=0x000D&dataPrefix=03&mask=FF), the actual manufacturer data of a SensorTag (`030000`) is allowed to exceed the length of the specified _filter_ for **manufacturer data** and **mask** (`03` and `FF`, respectively, in the screenshot):

[image]
closed 2025-01-27 by twyatt