JuulLabs/kable #840: Failed to establish connection with a ConcurrentModificationException

eygraber, 2025-01-20:
My app crashed shortly after starting to connect to a peripheral. 

```
java.util.ConcurrentModificationException
                                         	at java.util.LinkedHashMap$LinkedHashIterator.nextNode(LinkedHashMap.java:1061)
                                         	at java.util.LinkedHashMap$LinkedEntryIterator.next(LinkedHashMap.java:1096)
                                         	at java.util.LinkedHashMap$LinkedEntryIterator.next(LinkedHashMap.java:1093)
                                         	at com.juul.kable.Observers.onConnected(Observers.kt:134)
                                         	at com.juul.kable.Observers$onConnected$1.invokeSuspend(Unknown Source:14)
                                         	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                         	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:100)
                                         	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:586)
                                         	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:829)
                                         	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:717)
                                         	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:704)
                                         	Suppressed: kotlinx.coroutines.internal.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@476184a, Dispatchers.Default]
```

twyatt, 2025-01-20:
@eygraber can you try the following SNAPSHOT and see if it resolve the issue (it is based on [`0.35.0` release](https://github.com/JuulLabs/kable/releases/tag/0.35.0) w/ #841 cherry-picked).

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:kable-core:0.35.0+issue840-SNAPSHOT")
}
```

eygraber, 2025-01-20:
I haven't been able to repro it, but it makes sense that #841 would fix it because I found a bug in my code that was calling `observe` multiple times for the same characteristic very quickly.

PR #841 (2025-01-24): Create copy of `observations` in `Observers.onConnected`
Closes #840
closed 2025-01-24 by twyatt