JuulLabs/kable #843: Oob paring support ？

wang222222222, 2025-01-23:
Does this library support oob pairing?

twyatt, 2025-01-23:
Kable does not support OOB pairing. It sounds like it isn't supported by Android either. If Android supported it, then I could see Kable eventually getting support for it, but it doesn't seem like that will be any time soon. If ever?

twyatt, 2025-01-29:
If you have examples of other libraries support OOB, then I can use them as reference for implementing the functionality in Kable.

wang222222222, 2025-01-29:
Thank you for your reply, I can't find any support，Maybe this is a niche demand

twyatt, 2025-01-29:
> Maybe this is a niche demand

Sadly, seems that way. 😢 
I would've really liked such support; would be really nice to pair via NFC.

closed 2025-01-29 by twyatt