JuulLabs/kable #850: ScanError.toString-impl com.juul.kable.InternalException - Unsupported error code 100

mjordan-gpfw, 2025-01-31:
**Description**
We're seeing this exception thrown in production when trying to scan for ble devices.

```
com.juul.kable.scan.ScanError.toString-impl (ScanError.java:21)
com.juul.kable.BluetoothLeScannerAndroidScanner$advertisements$1$callback$1.onScanFailed (BluetoothLeScannerAndroidScanner.kt:69)
android.bluetooth.le.BluetoothLeScanner$1.run (BluetoothLeScanner.java:657)
android.os.Handler.handleCallback (Handler.java:958)
android.os.Handler.dispatchMessage (Handler.java:99)
android.os.Looper.loopOnce (Looper.java:224)
android.os.Looper.loop (Looper.java:318)
android.app.ActivityThread.main (ActivityThread.java:8777)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:561)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1013)
```

**Environment**
100% Xiaomi devices - 20+ different models. Mostly Redmi Note 13 Pro 5G, Poco X6 5G, and Xiaomi 13
Android OS Versions: 13 and 14 (our min is Android 11, but only see issues on 13/14)

twyatt, 2025-01-31:
Thanks for reporting!

I had hoped that the error codes defined in `ScanCallback` would be exhaustive:

| `ScanCallback` constant | Value |
|-|-|
| `SCAN_FAILED_ALREADY_STARTED` | `1` |
| `SCAN_FAILED_APPLICATION_REGISTRATION_FAILED` | `2` |
| `SCAN_FAILED_INTERNAL_ERROR` | `3` |
| `SCAN_FAILED_FEATURE_UNSUPPORTED` | `4` |
| `SCAN_FAILED_OUT_OF_HARDWARE_RESOURCES` | `5` |
| `SCAN_FAILED_SCANNING_TOO_FREQUENTLY` | `6` |

...but apparently not. Even worse, I looked for a while, and could not find in the Android source, where [`onScanFailed`](https://developer.android.com/reference/android/bluetooth/le/ScanCallback#onScanFailed(int)) is called from.

Internally, it appears that [Android just defaults to `Unknown`](https://cs.android.com/android/platform/superproject/+/android14-qpr3-release:frameworks/base/services/companion/java/com/android/server/companion/presence/BleCompanionDeviceScanner.java;l=393-394) as the display of any error besides what is known, hinting that any error code is possible. Which is super not helpful if you're trying to decide how you should handle such an error case.

Unless you can track down what error code `100` means, I'll just have to change Kable to not throw the `InternalException` and instead propagate an `IllegalStateException` with "Unknown" error code — not much else I think I can do from Kable to be more helpful, unfortunately. 😢

closed 2025-02-07 by twyatt

twyatt, 2025-02-13:
I'm realizing now, that `IllegalStateException` is intended for cases where the error case could've been avoided by a pre-condition check (e.g. BLE is unavailable and you don't check that ahead of time).

In this case though, it may warrant a separate exception class, to allow you to grab the error code value. Assuming there is some alternate action you can perform?

In other words, is error `100` preventable by a precondition check (can you check that something is in the expected state before starting the scan to prevent the error 100)? If there is an action you can take to prevent it, then `IllegalStateException` makes sense. But if there is nothing that can possibly be done, then it would likely make sense to throw as an `IOException`.

Basically, it is convention in Kable, that consumers should not catch `IllegalStateException`s because the app crash reporter should report that failure so that a bug in your app (consumer) code can be fixed. Whereas `IOException`s are out of control of the consumer (can't be checked ahead of time).

Anyways, let me know if you need me to make additional changes (if you need more information beyond just an `IllegalStateException` being thrown).