JuulLabs/kable #865: !!!! Remove the maxSdkVersion tags from the manifest permissions

bpappin, 2025-02-16:
**Please remove the maxSdkVersion tags from your manifest permissions.**

```xml
<uses-permission
        android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30"
        />
```

They overwrite the permissions in the applications that import kable-core so that any app built above 30 will exclude those permissions.
During manifest merge, the max version is carried into the host projects manifest, effectively removing those permissions for the app.

It took me many hours to track down what was going on, and figure out that it was kable including them.

twyatt, 2025-02-16:
I had originally included the Manifest permission to help make Kable more batteries-included experience (i.e. less/easier setup on the consumer side).

Although, similar to your issue, it has been previously reported how this can cause problems (#132).

It seems more and more that trying to make Kable easier to spin up in a project is creating a major footgun for others.

I'll revisit how Kable approaches this and consider removing all but the absolute essential permission declarations from Kable's Manifest(s) and instead providing documentation for recommended declarations.

bpappin, 2025-02-16:
I think it'll work, you need to remove the max version though. However, I would allow the calling app to add what they need, so they can manage it.

twyatt, 2025-02-16:
The `maxSdkVersion` declaration is the approach (verbatim) outlined in the [official docs](https://developer.android.com/develop/connectivity/bluetooth/bt-permissions).

Are the official docs providing incorrect setup guidelines?

bpappin, 2025-02-16:
Yes, that is how I would use them. When you needed them in a particular version and not in a later one.
My choice of example in the issue is probably not he best, I will adjust it now.

However in my case kable was including location permissions _(both fine and course)_ with the `maxSdkVersion` set to `30`.
What's happening is that during the build when the manifests all get merged together, the `maxSdkVersion=30` attribute overrides the rest of them in the app that didn't have the attribute. 
The result is that for an app running above sdk 30, those permissions are just not there, causing all permission checks to fail _(for location in my case)_. 

This is the offender:
https://github.com/JuulLabs/kable/blob/main/kable-core/src/androidMain/AndroidManifest.xml

They do have a purpose of course, but it would need to be very carefully considered, and it should definitely not be on everything.

I spent all day building a permission system for kmp, and it just would not work. 
I thought it should, but it was behaving like the permissions were not declared, yet I could see they were. 
In the end it took me taking apart the final product and examining the manifest to see what was happening, and then I had to go through all the libraries I was importing to figure out which one had the max version tags.

So, by all means use them as per the docs, but be very careful what you apply them to.
Right now, kable will break every app that need location permission running above sdk 30.

twyatt, 2025-02-17:
> However in my case kable was including location permissions (both fine and course) with the `maxSdkVersion` set to 30.
What's happening is that during the build when the manifests all get merged together, the `maxSdkVersion=30` attribute overrides the rest of them in the app that didn't have the attribute.
The result is that for an app running above sdk 30, those permissions are just not there, causing all permission checks to fail (for location in my case).

BLE permissions in Android are super complex/confusing, and even more so when you have to accommodate for many Android versions. The hope was that Kable would provide a [typical permission set for BLE apps that could be tweaked](https://github.com/JuulLabs/kable?tab=readme-ov-file#android-permissions) as necessary, making it more approachable for developers less familiar with Android BLE.

Perhaps that documentation for Kable (re: location permissions) is buried too far down the page?

I tried to strike a balance and provide an easy setup process for a majority of Kable consumers — assuming that many BLE apps won't also need location permission, and those that do: simply override Kable's declarations (via the specified `replace` attribute).

> I spent all day building a permission system for kmp, and it just would not work.
I thought it should, but it was behaving like the permissions were not declared, yet I could see they were.
In the end it took me taking apart the final product and examining the manifest to see what was happening, and then I had to go through all the libraries I was importing to figure out which one had the max version tags.

I'm sorry that so much time was lost trying to track down the issue. I had (perhaps wrongfully) assumed that Android Studio's "Merged Manifest" view (w/ "Jump to Declaration" for the the `maxSdkVersion`) would be enough for people to see that Kable was restricting the API versions for those permissions (if they had not been explicitly overridden via `replace`).

---

Ultimately, I do see how the pre-baked permissions can be problematic. Removing the pre-baked permissions will be helpful for some people (as you have clearly demonstrated) but will also break for others (that were leaning on the permissions that Kable provides).

I'll try to think of a sensible migration path to get to a place where Kable is defining less (as I do agree that is probably a better approach), but perhaps also be able to provide some sane/friendly defaults for those that find them useful.

bpappin, 2025-02-17:
The comment in that manifest for the location permissions indicates that you are trying to limit them to only include them for platforms below 31, however the outcome is not what you intended, the merge rules mean that they get excluded in platforms above 30.
This is happening based on the merge policies documented here, and should be testable:
https://developer.android.com/build/manage-manifests#merge_policies

This isn't really about `maxSdkVersion` being correct or not, it simply breaks applications that fire up on sdk 31+.
Personally, I'd like to keep using kable :)

Unfortunately we can only switch manifests on variants and the like, we can't do it like we do with resources. 
So, i see I few different options:

1. Simply remove the location permissions from the manifest, and document that they are needed if the app is running below a certain point.
2. put them into a separate library. Simply importing the library would apply the permissions.
3.  it might work to put them into a separate library and set the targetSdkVersion to 30, then simple depend on it from core. You will have to review the docs at: https://developer.android.com/build/manage-manifests#implicit_system_permissions and test if it works.
4. You could document that the host app must apply merge markers, but realistically the lower sdk versions are the ones that are going to become less and less prevalent. Most apps are going to be forced to add the markers. See: https://developer.android.com/build/manage-manifests#merge_rule_markers

twyatt, 2025-02-17:
I plan to go with a variation of 1 and 2. With item 2 being a dedicated permissions module/artifact.

PR #866 (2025-04-02): Provide default Android permissions via `kable-default-permissions` artifact
Closes #865
Closes #132

Removed Kable permission declarations and makes it the responsibility of the consumer to setup appropriate permissions. `kable-default-permissions` artifact provides sane defaults (same as previously) and can optionally be used by consumers.

Rendered markdown of the relevant updated documentation can be found [here](https://github.com/JuulLabs/kable/tree/twyatt/android-default-permissions?tab=readme-ov-file#android-permissions).
closed 2025-04-02 by twyatt