JuulLabs/kable #894: Manufacture code is different between iOS and Android for the same bluetooth device

xxfast, 2025-04-08:
Hi!

Thanks for the amazing library. It has been super helpful. 

I think I may have found a issue if the `Scanner`

I have this in my `commonMain`
```kotlin
val scanner: PlatformScanner = Scanner {} // scan everything

scanner
  .advertisements
  .collect { advertisement -> 
    logger?.d { """
      Advertisement: ${advertisement.identifier}
      Manufacturer code: ${advertisement.manufacturerData?.code}
      Manufacturer data: ${advertisement.manufacturerData?.data?.hex}
    """.trimIndent() 
    }
  }
```

On android logcat I see 
```
Advertisement: 02:57:29:16:C0:B0
Manufacturer code: 60902
Manufacturer data: ca 47 60 c6 58 fc ad 4e c5 2f 0e 33 e2 bc 2b 4f 12 (17)
```

Same device on ios gives 
```
Advertisement: b3cf2095-6955-aad7-b955-70e8f3ab2513
Manufacturer code: -4634
Manufacturer data: ca 47 60 c6 58 fc ad 4e c5 2f 0e 33 e2 bc 2b 4f 12 (17)
```

This seems like a bug? 🤔 

I'm using `kable-core@0.35.0`

twyatt, 2025-04-08:
Thanks for reporting. Looks like we aren't properly converting the data (that we read as a `short`) on iOS.

To illustrate the issue:

```kotlin
val code = 0xEDE6.toShort() // 60902
println(code.toInt()) // outputs -4634
```

I'll have this bug fixed in the next release.

closed 2025-04-10 by twyatt

xxfast, 2025-04-13:
Hi! Thanks for the fix

Any update as to when this is available? Are we waiting on #890?

twyatt, 2025-04-14:
> Any update as to when this is available?

Sorry about that, we ended up getting swamped on another project at work, so reviews didn't come as quick as I had expected.

> Are we waiting on https://github.com/JuulLabs/kable/pull/890?

Exactly right. Once I get another approval, I'll merge and release. I'm hoping to get that review tomorrow.

xxfast, 2025-04-14:
Thank you very much 🙏

PR #895 (2025-04-10): Fix handling of manufacturer code
Closes #894