JuulLabs/kable #898: Use Kotlin constants to list Bluetooth SIG assigned numbers

chungchungdev, 2025-05-05:
[Creating 128-bit UUIDs from Bluetooth assigned numbers using the String-to-Int map](https://github.com/JuulLabs/kable/blob/main/kable-core/src/commonMain/kotlin/Uuid.kt) is a bit inconvenient. Would you consider listing the assigned numbers as Kotlin constants? Having autocompletion saves some effort.
I made a [file](https://github.com/chungchungdev/kotlin-bluetoothsig-assigned-numbers/blob/master/generated/AssignedNumbers.kt) that lists all assigned numbers of service, characteristic and descriptor as Kotlin constants.

twyatt, 2025-05-06:
Yes, I think having Kotlin constants would be much better. Thanks for providing the link to a possible implementation. I'll look into this soon.

chungchungdev, 2025-05-06:
Sure, I added an Apache-2.0 license.

twyatt, 2025-05-06:
Awesome! Thank you! I'm traveling at the moment, but will look into this when I get back later this week.