JuulLabs/kable #904: [iOS] Expose `setDesiredConnectionLatency(_:for:)`

yuroyami, 2025-05-17:
Similar to Android's `requestConnectionPriority()`, iOS provides [`setDesiredConnectionLatency(_:for:)`](https://developer.apple.com/documentation/corebluetooth/cbperipheralmanager/1393277-setdesiredconnectionlatency) which allows for more focused energy-vs-performance control. However, this functionality is not currently exposed in the framework (that I know of). I believe it should be commonized across platforms, creating a common interface that maps to the appropriate Android and iOS implementations since they essentially serve the same purpose.

twyatt, 2025-05-17:
`setDesiredConnectionLatency` is a method on `CBPeripheralManager`, so is intended for the "server" size of the connection.
Unfortunately, I don't believe there is a way to set the connection latency with Core Bluetooth from the client (peripheral) side of the connection (for which Kable is providing the API for).

yuroyami, 2025-05-17:
Makes sense. Thanks for the clarification!

closed 2025-05-17 by yuroyami