JuulLabs/kable #912: onSubscriptionAction is not called for subsequent subscribers if observation is already started

renaudcerrato, 2025-05-24:
The documentation of `Peripheral.observe()` states the following about `onSubscription`:

```
- It will be executed whenever [connection][connect] is established (while the returned [Flow] is active); and
- It will be executed _after_ the observation is spun up (i.e. after enabling notifications or indications)
 
If multiple [observations][observe] are created for the same [characteristic] but with different [onSubscription]
     * actions, then the [onSubscription] actions will be executed in the order in which the returned [Flow]s are
     * subscribed to.
 
```

According to my understanding of the current documentation I'm expecting `onSubscription` to be called for every subscribers if connected.

However, I've noticed that `onSubscription` is not called for subsequent subscribers if the observation already started, which seems confirmed by looking at https://github.com/JuulLabs/kable/blob/main/kable-core/src/commonMain/kotlin/Observation.kt#L39 :

```kotlin
    suspend fun onSubscription(action: OnSubscriptionAction) = mutex.withLock {
        subscribers += action
        val shouldStartObservation = !didStartObservation && subscribers.isNotEmpty() && isConnected
        if (shouldStartObservation) {
            suppressNotConnectedException {
                startObservation()
                action() // <---- here, called only if didStartObservation was false
            }
        }
    }
```

Is that the expected behavior? A possible fix would be (not yet tested):

```kotlin
    suspend fun onSubscription(action: OnSubscriptionAction) = mutex.withLock {
        subscribers += action
        if (isConnected) {
            if (!didStartObservation) {
                suppressNotConnectedException {
                    startObservation()
                    action()
                }
            } else {
                // Observation already started, just call the new subscriber's action
                suppressNotConnectedException {
                    action()
                }
            }
        }
    }
```

Let me know If I missed something or if you need a PR.

twyatt, 2025-05-26:
The use-case for the `onSubscription` is where you have a peripheral that needs an additional write when enabling an observation (e.g. the peripheral needs a write to a specific characteristic in addition to the standard CCCD mechanics of BLE).

The `onSubscription` allows that write to occur alongside the "enable observation" action _without_ missing a potential response from peripheral.

Let me know your use-case and the current API may not account for it. I believe (from your suggested code changes), that simply using [`onStart`](https://kotlinlang.org/api/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/on-start.html) on the flow returned from `observe` might work for you? As I'm assuming you're not interested in a _response_ message from your peripheral on observation?

twyatt, 2025-05-26:
> I'm sending a command during onSubscription, while the subscriber wait for the matching reply (using filters). I'm basically registering a new (one-shot) subscriber for every command to be sent.

This is exactly what we do w/ some of our internal projects, for example, here is a slightly abbreviated version of what we use:

```kotlin
/** Send CoAP message and wait for the corresponding CoAP response. */
private suspend fun sendForResponse(request: Message.Tcp): Message.Tcp {
    val response = incoming
        .onSubscription { send(request) } // `send` writes to the outgoing characteristic
        .first { response -> response.token == request.token }
    // ..
}
```

> I'm afraid that using `onStart()` to send my command would miss the notification since `onStart()` may be called before the subscription is active.

Ya, for the use-case you've described, `onStart` is not a good choice (could miss a response).
You can either use `onSubscription` from a `SharedFlow` (i.e. using `shareIn` on `observe`) or `async` read of the incoming flow with `start = UNDISPATCHED` (there was some discussion re: this approach in https://github.com/JuulLabs/kable/issues/794).

renaudcerrato, 2025-05-26:
Great suggestions, will give them a try! Thanks!

closed 2025-05-26 by renaudcerrato

renaudcerrato, 2025-05-27:
The `UNDISPATCHED` approach works like a charm. Thanks again!

```kotlin
 return peripheral.connect().async {

            val deferred = async(start = UNDISPATCHED) {
                notifications.filterIsInstance<T>().first()
            }

            request(type)

            deferred.await()
        }.await()
```