JuulLabs/kable #915: Scanning is broken on android when upgrading to version 0.37.1

rawphl, 2025-05-29:
In version 0.36.0, this code worked to scan for Adafruit Bluefruit LE devices:

`
    private val scanner: Scanner<Advertisement> = Scanner {
        filters {
            match {
                name = Filter.Name.Exact(ADAFRUIT_DEVICE_NAME) // "Adafruit Bluefruit LE"
            }
        }
        logging {
            level = Logging.Level.Events
        }
    }
`

When upgrading to 0.37.1, the scanning doesn't find any device at all, regardless of the filters. I tried:

- No filters at all
- Filtering by service (UART_SERVICE_UUID = "6e400001-b5a3-f393-e0a9-e50e24dcca9e")
- Filtering by name prefix

This issue was introduced in #890. I've read through the changes, but can't spot the issue. For now, I will stay on 0.36.0, but I am still curious if this is a bug or if I missunderstand how the filters have to be specified.

twyatt, 2025-05-29:
It is possible a bug was introduced, but the fact that you're not finding any devices at all with "no filters at all" is surprising (as I would be surprised if #890 affected that configuration).

Could you provide the logs that are produced when executing your scan?

twyatt, 2025-05-29:
The sample code you provided looks good. It is bizarre that your no-filter scanner doesn't find any devices, since the log is explicitly stating that Kable is "Starting scan without filters". So, the filter list that Kable is using is empty.

Are you sure that same code is working with the 0.36.0 of Kable? I'm wondering if maybe the phone you're using is just in a weird state and a reboot of the phone might fix things?