JuulLabs/kable #959: Inconsistent connection state when bluetooth is turned off

twyatt, 2025-07-17:
Hello, I considered opening an issue for this, but I'm not sure if I'm using the library wrong. So I opened a discussion instead.

My question is : On iOS, given a peripheral with an active connection, should turning off the Bluetooth radio result in the `peripheral.state` flow emitting a `Disconnected` state?

In my app I never see this. On Android it works a 100% of the time but on iOS I see the following sequence of events in the Kable logs. Disconnection always times out after 5 seconds; and the `Disconnected` state is never emitted. Since things work as expected on Android, I'm assuming this might be something I'm doing wrong.

```
I/Kable: <SOME-UUID> Bluetooth powered off
  state: 4
V/Kable/Connection: <SOME-UUID> Waiting for connection tasks to complete
D/Kable/Connection: <SOME-UUID> Disconnecting
V/Kable/Connection: <SOME-UUID> cancelPeripheralConnection
API MISUSE: <CBCentralManager: 0x30259c000> can only accept this command while in the powered on state
W/Kable/Connection: <SOME-UUID> Timed out after 5s waiting for disconnect
D/Kable/Connection: <SOME-UUID> Closing
kotlin.coroutines.cancellation.CancellationException: com.juul.kable.NotConnectedException: Disconnect requested
    at 0   MyAppName.debug.dylib      0x105570eeb        kfun:kotlin.Throwable#<init>(kotlin.Throwable?){} + 243 
    at 1   MyAppName.debug.dylib      0x10556b0a7        kfun:kotlin.Exception#<init>(kotlin.Throwable?){} + 95 
    at 2   MyAppName.debug.dylib      0x10556b277        kfun:kotlin.RuntimeException#<init>(kotlin.Throwable?){} + 95 
    at 3   MyAppName.debug.dylib      0x10556b73b        kfun:kotlin.IllegalStateException#<init>(kotlin.Throwable?){} + 95 
    at 4   MyAppName.debug.dylib      0x105577fe3        kfun:kotlin.coroutines.cancellation.CancellationException#<init>(kotlin.Throwable?){} + 95 
    at 5   MyAppName.debug.dylib      0x105f81283        kfun:com.juul.kable.CBPeripheralCoreBluetoothPeripheral#disconnect#suspend(kotlin.coroutines.Continuation<kotlin.
I/Kable/Connection: <SOME-UUID> Closed
```

From my reading of the source, it looks like there are 2 components at play here
1. [`onBluetoothPoweredOff`](https://github.com/JuulLabs/kable/blob/cbe69ab28a62c858c149436b9006084dbd2a801d/kable-core/src/appleMain/kotlin/CBPeripheralCoreBluetoothPeripheral.kt#L77) in `CBPeripheralCoreBluetoothPeripheral.kt`
2. [`onDispose`](https://github.com/JuulLabs/kable/blob/cbe69ab28a62c858c149436b9006084dbd2a801d/kable-core/src/appleMain/kotlin/Connection.kt#L70) in `Connection.kt`.

It looks like for some reason, when `onBluetoothPoweredOff` happens, then `disconnect` is also being called. My guess is that `CBCentralManager.cancelPeripheralConnection` is only allowed to be called when the Bluetooth radio is powered on and that explains the rest of the log output.

However, what I don't understand is: Why is `onBluetoothPoweredOff` eventually calling `disconnect` in Perpiheral.kt? Could it be something do with the coroutine scope that I'm using to collect the `peripheral.state` flow?

Does anyone have any ideas?

_Originally posted by @curioustechizen in https://github.com/JuulLabs/kable/discussions/954_

twyatt, 2025-07-30:
@curioustechizen were you only seeing this issue when disabling bluetooth from the control center (and not when disabling from settings)?

I'm wondering if Kable needs to account for the iOS bug similar to https://github.com/JuulLabs/topaz/pull/143.

curioustechizen, 2025-07-31:
I've only been testing the "Disable BT from the control center" case all this while. It never occurred to me that the behaviour would be different when disabling from settings.

But yes I see the exact behaviour that you describe:
- Disable Bluetooth from settings: Everything works as expected, I receive the `Disconnected` state
- Disable Bluetooth from Control Center: The issue described in the original bug is seen

closed 2025-08-05 by twyatt