JuulLabs/kable #1005: Connection state transitions directly to CONNECTED after GATT_ERROR during onServicesDiscovered.

mobilestromer-ctrl, 2025-08-19:
Hello, what we saw is that the peripheral connection state transitions to CONNECTED before DISCONNECTED (on Android side) while a GATT_ERROR occurs during the handshake from onServicesDiscovered. What we would expect is that it transitions to DISCONNECTING/DISCONNECTED instead, so we should reliably know if the connection was properly established. Thank you!

mobilestromer-ctrl, 2025-08-20:
We are using 0.39.2, this happens in the context of a kmm project, with the ble functionality being shared, but only when running it on Android devices.

mobilestromer-ctrl, 2025-08-26:
I understand that bonding is not supported, but we expect a DISCONNECTING/DISCONNECTED state to be returned when an error is thrown during characteristic observations stage.

twyatt, 2025-08-26:
> we expect a DISCONNECTING/DISCONNECTED state to be returned when an error is thrown during characteristic observations stage

That's a good point. I'll try to find some time to look into it.

mobilestromer-ctrl, 2025-09-09:
Hello, any news on the matter?